riscv-non-isa/riscv-brs #4: BRS Recipe Naming

andreiw, 2023-02-06:
The original https://docs.google.com/document/d/1X0TSbheEJjRGWhG2nzUnfIl_QcHWSuvd/edit# document called the recipes as "HI" and "VI", standing for "Horizontally-Integrated" and "Vertically-Integrated".

Mark didn't like the names, and we should think if better ones exist.

https://github.com/riscv-non-isa/riscv-os-a-see/pull/3 uses BSR-H and BSR-V as a stop-gap. Personally, I'm a fan of using Vertical and Horizontal, perhaps clarifying that we're talking about standards. There is a precedent elsewhere - https://legalbeagle.com/13659495-according-to-osha-what-is-the-difference-between-vertical-horizontal-standards.html

To help thinking of alternate names, here is a rough break-down of H vs V:

"Horizontal":
- Enables wide set of OSVs, OEMs, IHV and SiPs to interop in a compatible manner.
- "team effort": one company builds Si, another packages into a board, another packages into box, yet another writes OS, yet another writes apps, yet another one sells and provides services
- servers, desktops, laptops

"Vertical":
- Purpose-built hardware + app + service
- One vendor controls everything from hardware and firmware to application/service and sales
- Examples: mobiles, embedded, cloud

mark-riscv, 2023-02-08:
how about something like "shared" or "community" for HI and "custom" or "proprietary" for VI?

adurbin-rivos, 2023-02-13:
> how about something like "shared" or "community" for HI and "custom" or "proprietary" for VI?

I'm not sure we should be adding that much connotation to the various solution recipes.

adurbin-rivos, 2023-02-13:
I'm not against horizontal vs vertical as long as we write up the definition/motivation for the names. I think they can work. However, if those don't work we may want to be generic as possible. Recipe A, Recipe B, etc. However, I don't think that'll help people remember the distinction between the two.

andreiw, 2023-02-13:
> I'm not against horizontal vs vertical as long as we write up the definition/motivation for the names. I think they can work. However, if those don't work we may want to be generic as possible. [...]

Yeah I am not a fan of A vs B. A previous project I was on did that (customer A and customer B) and it was confusing. It's bound to get even more confusing with multiple orgs involved.

andreiw, 2023-02-13:
> how about something like "shared" or "community" for HI and "custom" or "proprietary" for VI?

It's fair to argue that "proprietary" is somewhat in the eye of the beholder. The U-Boot + DT style Arm SBC ecosystem sure considers itself to be an open community, for example. The VI solutions /can/ be more bespoke and don't factor in interoperability.

Interoperable vs Bespoke? Doesn't quite roll off the tongue, though.

andreiw, 2023-02-14:
Maybe I and B aren't that bad. BRS-I and BRS-B?

closed 2023-03-06 by adurbin-rivos