riscv-non-isa/riscv-brs #12: Dependency: ACPI 6.6: APEI

andreiw, 2023-03-29:
Does anything RISC-V specific need to be added to e.g. GHESv2 or elsewhere.

vlsunil, 2023-04-06:
Yes, both ACPI and UEFI spec needs some updates for RISC-V.

hschauhan, 2023-04-06:
Yes, there is a need to define an ID in SectionType of Section Descriptor. This wold be to differentiate between Error from different type of sources. There is Generic/IA32/IA64/ARM processors defined. We will need one for RISCV. Also, Ventana is working on an high priority software events specification. This would need identification number to differentiate between SDEI of ARM and SSE of RISCV. We at Ventana are working on a proposal for SSE specfication and RAS software architecture.

andreiw, 2023-04-12:
Need to clearly understand what parts of APEI can be implemented without depending on the ongoing RAS work (i.e. apply to existing hardware). Can fw-first error handling be made a requirement for BRS or not or if it is conditional on certain HW features being implemented.

andreiw, 2024-04-11:
RAS is not currently in scope.

closed 2024-04-11 by andreiw