riscv-non-isa/riscv-brs #18: Dependency: RIMT (IOMMU) ACPI table

andreiw, 2023-03-30:


andreiw, 2025-01-23:
Now that we're ratified, let's revisit this for v2.

andreiw, 2025-01-23:
See https://github.com/riscv-non-isa/riscv-server-platform/issues/2. RIMT requirement is being put into the Server Platform spec for now but it certainly would be well-placed in the BRS. Of course in the BRS it would need to be made conditional on the presence of an IOMMU, but there is a precedent for that (with the TPM).

avpatel, 2025-03-19:
The BRS v1.0 has already completed freeze vote at TG-level and HC-level so this will be an enhancement for BRS v2.0

ved-rivos, 2025-03-19:
It could be addressed as part of the public review feedback - if it makes sense to include it in 1.0.

andreiw, 2026-10-29:
Fixed in db8537a42f75c430e48e4720bf30185f35d0262d by @vlsunil

closed 2026-10-29 by andreiw