riscv-non-isa/riscv-brs #21: Security requirements?

andreiw, 2023-03-30:
Do we consider these part of the BRS-I or should this be a separate document like BBRS?

adurbin-rivos, 2023-03-31:
What security aspects are you thinking should be covered?

andreiw, 2023-04-12:
To paraphrase the Arm BBRS (https://developer.arm.com/documentation/den0107/latest/)

Platform requirements for BRS-based systems that enable standard, suitably built
operating systems to seamlessly use standard security interfaces. These interfaces include the following security
related functionality:
• UEFI authenticated variables
• UEFI secure boot
• UEFI secure firmware update using Update Capsules
• TPMs and measured boot

The Arm BBRS also covers platform reset attacks, and guidelines for implementing.

Maybe don't have to overthink this and it can be first defined under a server platform spec and then factored out. Or it could be an optional addendum to BRS.

Thoughts?

andreiw, 2024-04-11:
Need to revisit this, esp wrt https://github.com/riscv-non-isa/riscv-brs/issues/136 and https://github.com/riscv-non-isa/riscv-brs/issues/135