riscv-non-isa/riscv-brs #22: Dependency: ACPI: Define _CCA behavior for RISC-V

andreiw, 2023-03-30:
Recommend this as ASWG change to mandate _CCA on any RISC-V platform or force a specific meaning when _CCA is not present.

andreiw, 2023-03-30:
Should we document _CCA to follow X64/AARCH64 semantics? Defaulting to cache coherence when absent?

adurbin-rivos, 2023-03-31:
Probably good for a default in ACPI-land. A particular RISC-V Platform may have stronger expectations for all devices in a system to be cache coherent.

paul-walmsley-sifive, 2023-04-10:
> Recommend this as ASWG change to mandate _CCA on any RISC-V platform or force a specific meaning when _CCA is not present.

Agreed with either of these two approaches

andreiw, 2023-08-24:
Filed https://mantis.uefi.org/mantis/view.php?id=2407. On RISCV, if the \_CCA object is not supplied, the OSPM will assume the devices are hardware cache coherent.

closed 2024-04-11 by andreiw