riscv-non-isa/riscv-brs #23: Do we need a LinuxBoot recipe?

andreiw, 2023-03-30:
Comments on the Illustrative doc:

* The ARM LBBR-v1 profile doesn't appear to mandate any UEFI boot services.  Should we mandate all of these for all OS-A SEE profiles?
* Where do LinuxBoot requirements fit in? If BRS-I, the regular server class requirements do not hold good. Should it be separate recipe itself like ARM did?

I could see LinuxBoot to be a case of BRS-B (what is called VI here). If folks disagree because that's too open-ended, we can add a separate recipe

paul-walmsley-sifive, 2023-04-03:
I think I may have been the one that originally commented about this.   Seems like we can put it off until a future revision of the specification or until someone who has a strong interest in this joins the group.

andreiw, 2023-04-12:
Sounds good. Closing for now.

closed 2023-04-12 by andreiw