riscv-non-isa/riscv-brs #24: Define AML device for SPCR type 0x12 devices

andreiw, 2023-03-30:
Type 0x12 can describe NS16550-like UARTs that cannot be described by PNP0501 - the e.g. register stride, clock frequency. And there are other "tunables" which may be interesting to the OS driver. This should have it's own ACPIxxxx _CID.

Can model this on https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/serial/8250.yaml.

These seem like a good set. Should these use DSD DT?

- clock-frequency
- reg-offset
- reg-shift
- reg-io-width
- fifo-size

adurbin-rivos, 2023-03-31:
'DSD DT' - what do you mean by this? device specific data and have that under a specific ACPIxxxx device name?

paul-walmsley-sifive, 2023-04-03:
> 'DSD DT' - what do you mean by this? device specific data and have that under a specific ACPIxxxx device name?

He's probably referring to the practice of embedding DT properties into the _DSD entries, i.e., https://docs.kernel.org/6.2/firmware-guide/acpi/dsd/graph.html

adurbin-rivos, 2023-04-04:
Yes, that was my question confirming if DSD DT meant that. I assumed so, but I wanted to be sure.

andreiw, 2023-04-12:
Yes, Paul is correct.

andreiw, 2024-04-11:
This would be nice to have but ideally is code-first and fueled by someone who needs it

andreiw, 2024-04-15:
Also need to add the rscv- Known Device Property Prefixex

closed 2024-04-18 by andreiw