riscv-non-isa/riscv-brs #28: Support for harts with asymmetric perf

andreiw, 2023-04-27:
Although the server HW TG is not spun up yet, the current WIP draft allows for "turbo boost max"-like implementations, where certain cores could support an even higher perf state. And this could mean P-core/E-core (bit.LITTLE-like) implementations too.

Need to figure out fw implications, including but not limited to:
- reporting capabilities per hart (static via RHCT or via AML?)
- Default boot state (do all cores boot at the same performance level? is the "boot" cpu the lowest performing one or the highest-performing one?)

The goal here is to cross the Ts and dot the I's so we don't end up in a world where someone builds a system that is compatible to the HW spec but somehow isn't fully expressible via the FW one.

xypron, 2024-02-14:
@andreiw 
Your ideas concerning heterogeneous harts seem to be included in commit e022aebf9c7e ("Expand recipes a bit going into further requirements for BRS-I.").

Can this issue be closed?

andreiw, 2024-04-11:
I think so.

closed 2024-04-11 by andreiw