riscv-non-isa/riscv-brs #34: Definition and usage of the term "system" is confusing and inconsistent

darius-bluespec, 2023-05-16:
The two sentences of the glossary definition for _system_ directly
contradict each other.  The first sentence says a system is both a
_platform_ and _software_; the second sentence contradictingly
says it is both of those things but also only one of those things.

Elsewhere in the document, _system_, when used independent of
compound terms (i.e., not "operating system", etc.) frequently refers
to **only** that which implements the BRS and **not**, for example, an
operating system or user application.

I would suggest completely replacing use of the term system with more
specific and appropriate terms, but I do understand this approach may
not preferred by others.

adurbin-rivos, 2023-05-17:
The 2nd sentence in question is:

> A system can be thought of both as a logical construct (e.g. a software stack) or physical construct (e.g. a notebook, a desktop, a server, a network switch, etc).

So if we remove that and keep the 1st sentence:

> A system is the entirety of a computing entity, including all elements in a platform (hardware, firmware) and software (hypervisor, operating system, user applications, user data). 

That should be sufficient? Then independent usage of the 'system' would be consistent?

I don't quite understand the suggestion of no using a consistent term throughout if the definition matches. It can be left to the reader how to run w/ the definition. I don't think it's that far of a leap to suggest a reader can do that.

darius-bluespec, 2023-05-18:
> That should be sufficient? Then independent usage of the 'system' would be consistent?

Now you have constructed a definition that, while consistent with itself, contradicts the majority of usage in the document.  For example, a _system_ by this definition is not the kind of system that _system_ _vendors_ are vending.  Nor can you talk about software, obtained separately, running on the system, because the system includes that software.

> I don't quite understand the suggestion of no using a consistent term throughout if the definition matches. It can be left to the reader how to run w/ the definition. I don't think it's that far of a leap to suggest a reader can do that.

The challenge is whether the authors can in fact create a definition that is consistent with usage.  This is exactly the problem; that there are multiple, commonly used, colloquial usages of the term _system_ in this context, and it is difficult to decide on and be consistent with only one.

andreiw, 2024-04-11:
Lets go with the newer https://github.com/riscv-non-isa/riscv-brs/issues/125 issue and see a PR.

closed 2024-04-11 by andreiw