riscv-non-isa/riscv-brs #36: The relationship between recipes and sections is not clear

darius-bluespec, 2023-05-16:
Recipes should be composed of **sections** in the BRS, not external
specifications.  The current section organization is confusing with
respect to what is and is not a requirement for a particular recipe.
For example, the ACPI section refers to many requirements as
mandatory, but the intention is that entire section is ignored (i.e.,
not mandatory) if not called for in a recipe, such as BRS-B, where
ACPI is intended to be optional.

adurbin-rivos, 2023-05-17:
Organizationally, do you think we should split into 'common', brs-i, and brs-b? Or orient around brs-i and brs-b w/ those other aspects underneath them? The latter may lead to some duplication, but we could figure out something there.

andreiw, 2024-04-11:
I think we ended up doing exactly what this ticket proposes.

closed 2024-04-11 by andreiw

darius-bluespec, 2024-04-18:
No you did not.