riscv-non-isa/riscv-brs #37: Some optional requirements are written in an inflexible way

darius-bluespec, 2023-05-16:
Requirements are stated in a way that precludes vendor specific
behavior in certain cases.

For example, "the PCI Memory-mapped Configuration Space (MCFG) table
must only be present if and only if compatible non-hot-removable PCIe
segments are made available to the OS."  Thus, implementors have the
option of (1) not implementing that feature, (2) implementing it
exactly as described, or (3) being out of compliance with the entire
specification.  There is no latitude to comply with the BRS and
provide that feature in a vendor specific way.

Contrast to the way that ISA extensions are defined, considering for
example floating point (F&D).  An implementation may (1) not implement
floating point, (2) implement floating point as described by the F&D
extensions, or (3) implement floating point in any other way.  All
three options are compliant with the ISA specification.  The only
prohibition is that option (3) cannot assert that it implements the
F&D extensions.

adurbin-rivos, 2023-05-17:
I think the current description could be better clarified if we're tying things to the PCIe FW spec. I see the citations, but we aren't calling out those specific requirements in what is written now. I suggest we say compliant with PCIe FW spec, basically.

That said, I think BRS is supposed to contain best practices based on experience. It is true there were issues relating to MCFG on certain systems. As such, I think we should call out the requirements to not allow for missteps.

darius-bluespec, 2023-05-18:
One person's "abuse" and "OS hack" is another person's "vendor specific behavior".  I don't particularly care about this specific case, but I am concerned about setting a precedent of overly constraining optional requirements and limiting flexibility for vendor specific behavior.  It seems excessive in the BRS Specification; perhaps more appropriate in a platform specification.  The problem with unnecessarily overconstraining things is that the requirements get ignored and the specification loses credibility.

andreiw, 2024-04-11:
> I think the current description could be better clarified if we're tying things to the PCIe FW spec. I see the citations, but we aren't calling out those specific requirements in what is written [...]

I like this approach.

Hijacking this issue to clean up the MCFG issue you highlighted... if you see other instances where we are being needlessly specific instead of deferring to the right spec, let us know.

andreiw, 2024-04-11:
> One person's "abuse" and "OS hack" is another person's "vendor specific behavior". I don't particularly care about this specific case, but I am concerned about setting a precedent of overly [...]

BRS-I is very much about avoiding vendor specific behavior that violates an industry spec. You can still write your own custom code. Just don't violate the PCIFW by abusing a specific ACPI table reserved for compliant implementations. It's worth highlighting this as a requirement...while you'd think it's common sense, what happened with AArch64 and ACPI support begs to differ.

closed 2024-04-11 by andreiw