riscv-non-isa/riscv-brs #39: UEFI spec version vs a minimal subset

atishp04, 2023-05-22:
UEFI spec section 2.6 lists the various boot/runtime services and platform specific elements. If we mandate a specific version in BRS spec, it may mean that all of the things specified in the spec has to be implemented by the firmware.

For EDK2, it probably won't be an issue but it makes harder for any other firmware/boot loader to comply.

There are many platform specific requirements which may not be required.
Example:
If a platform includes USB bus support, then EFI_USB2_HC_PROTOCOL and EFI_USB_IO_PROTOCOL
must be implemented. An external device can support USB by producing a USB Host Controller Protocol.

If a platform includes an NVM Express controller, then EFI_NVM_EXPRESS_PASS_THRU_PROTOCOL must
be implemented. 

I think we should add a UEFI [conformance profile](https://uefi.org/sites/default/files/resources/UEFI-Conf_Tables-WP-July-2022v2.pdf) for each recipe to make it easier.  EBBR already defines a one for the its own UEFI subset.

adurbin-rivos, 2023-05-24:
Thanks for the pointer. Presumably, we would put a profile in that would indicate minimum support. Essentially 2.6.1 Required Elements. And punt on the following as optional:

- 2.6.2 Platform-Specific Elements
- 2.6.3 Driver-Specific Elements

Looks like appropriate section for conformance profiles is 4.6.5 EFI_CONFORMANCE_PROFILE_TABLE. There's only currently 1 listed (to adhere to all requirements encompassed in  2.6 requirements).

adurbin-rivos, 2023-05-24:
EBBR publishes its own GUID: https://github.com/ARM-software/ebbr/blob/main/source/chapter2-uefi.rst#efi-conformance-profile-table

I missed the original suggestion about associating a conformance profile for each recipe. Question I have is what happens when we extend it? Looks like EBBR uses semantic versioning and any value in the patch level of the version number is deemed compliant.

adurbin-rivos, 2023-05-24:
EBBR decided to go down the route of listing required pieces as well as exceptions to those other line items.

atishp04, 2023-05-25:
> 

> EBBR decided to go down the route of listing required pieces as well as exceptions to those other line items.

Yeah as few of those are largely unused in modern system. If we can relax the requirement section in UEFI spec itself, we won't have to do anything. I am not sure if that's something feasible keeping our timeline.

andreiw, 2023-06-20:
> > EBBR publishes its own GUID: https://github.com/ARM-software/ebbr/blob/main/source/chapter2-uefi.rst#efi-conformance-profile-table
> > I missed the original suggestion about associating a conformance profile for each recipe. Question I have is what happens when we extend it? Looks like EBBR uses semantic versioning and any value in the patch level of the version number is deemed compliant.
> 
> For BRS-B, follow conformance profile defined by EFI_CONFORMANCE_PROFILE_EBBR_2_1_GUID For BRS-I, define a new conformance profile with a 2.6.1 (required) and subset of 2.6.2 & 2.6.3. We also need to define required/optional bootime/runtime services. [ARM base boot requirement](https://developer.arm.com/documentation/den0044/f) can be a good starting point for this.

The original draft BRS doc (https://docs.google.com/document/d/1X0TSbheEJjRGWhG2nzUnfIl_QcHWSuvd/edit#heading=h.38op2zyxb220) talked about defining a conformance profile (EFI_CONFORMANCE_PROFILE_OS_A_SEE_1_0_GUID).

...I think a conformance profile is a good idea. We can refer to the the laundry list in SBBR as a start (I kinda duplicated that in the draft doc) and define the subset of that for BRS-I.

Like I mentioned, this doc needs a BRS-I-S or BRS-SI defined as well (the server fw reqs on top of BRS-I), and that ought to have a separate conformance profile that is a superset of the BRS-I one.

andreiw, 2023-08-25:
So the question is 2.6.2 Platform-Specific Elements (EFI_CONFORMANCE_PROFILES_UEFI_SPEC_GUID) too restrictive for the BRS-I (not BRS-B aka EBBR) use case?

I took a long hard look at 2.6.2 and it's common sense stuff for a non-embedded implementation.

andreiw, 2023-10-18:
Complete?

closed 2023-10-18 by andreiw