riscv-non-isa/riscv-brs #45: Secondary Core Boot chapter

atishp04, 2023-06-09:
Sorry if this was discussed earlier in the TG that I did not attend. I couldn't find any context about this.
Is this supposed to be a booting guide style document  that specifies the booting vs non booting hart state, assumptions ?

We are working on a booting guide doc for Linux. We can refer to that here if that is the intention.

andreiw, 2023-08-22:
Just to come back to this, are you thinking there should be more in the secondary hart boot chapter? If so, that should be here for sure (not sure referring to a Linux-specific booting guide doc is really a good idea).

atishp04, 2023-08-22:
@andreiw : I was not sure what was the intention earlier. However, after discussing with Aaron, the chapter is updated.
https://github.com/riscv-non-isa/riscv-brs/blob/main/secondary.adoc

Let me know if any more details needs to be there.

andreiw, 2023-08-22:
I have a proposed change in https://github.com/riscv-non-isa/riscv-brs/pull/65