riscv-non-isa/riscv-brs #52: Confusing language around SUSP and reset

andreiw, 2023-08-02:
https://github.com/riscv-non-isa/riscv-brs/blob/8318e6b79701fcc1551302ed31a27b24b322e00d/sbi.adoc?plain=1#L21

SUSP has no UEFI RT equivalent, so it doesn't make sense to guide readers to prefer a UEFI interface here. Larger question is why is it being mandated? I can imagine devices where suspend-to-ram is not a product requirement.

I think we talked about mandating SRST, so I don't want to revisit that, but it is a _requirement_ (not a preference or suggestions) that OSes use the UEFI RT system reset call over SBI.

adurbin-rivos, 2023-08-07:
And the SUSP being mandatory would be the other issue to address that isn't sorted in https://github.com/riscv-non-isa/riscv-brs/pull/55/ ? 

@atishp04 fyi

adurbin-rivos, 2023-08-07:
We discussed this topic in BRS meeingt today: https://docs.google.com/presentation/d/1bWS_YUZAZwwDDgHOWf2S1vyoP_Eh7y6090-NjoJQTSo/edit#slide=id.g21d5651495d_0_17

Suspend shouldn't be mandatory, but we don't have the ACPI framework in place to allow arbitrary sbi calls?

andreiw, 2023-08-21:
Reopen since SUSP is still mentioned in the doc. I'll send a separate pull request to deal with that...

andreiw, 2023-08-22:
Okay, so the issue is resolved.

Let's now rename it so it instead tracks any gaps in ACPI and FFH around exposing suspend capability.

closed 2023-08-22 by andreiw