riscv-non-isa/riscv-brs #71: Error handling (kernel vs firmware first)

andreiw, 2023-09-07:
Current language for BRS-I mentions firmware-first (APEI) handling. We need to close on whether that's really doable for v1 of the BRS or whether we should just remove any requirement on error handling implementation for now.

dhaval-rivos, 2023-09-09:
Let us discuss in BRS but we have spec WIP already as part of RAS POC.

adurbin-rivos, 2023-09-11:
Sounds like there are in-flight changes to ACPI and UEFI (CPER) to be tracked in PRS.  @dhaval-rivos indicating we need these changes to have a more functional APEI. Time frame on these changes proposals?

andreiw, 2023-09-14:
This is what I got back from Himanshu at Ventana: APEI doesn't require any additional hardware capabilities. 

So there's some disagreement here. To be investigated.

dhaval-rivos, 2023-09-14:
Hi Andrei, I am not exactly sure of what Himanshu meant, but unless I misunderstood the question, the let me provide specific example why I said APEI/CPER changes are tied to HW. i.e. we have notification method for RAS events. This notification method is HW dependent (RAS exception). Similarly CPER table contains information that comes from RV HW. So if the question was we could define APEI/CPER independent of RERI spec being defined, I still think there is dependency.

avpatel, 2023-09-14:
List of things required for APEI/CPER are:
* RAS Local IRQs which already standardized/ratified by AIA spec.
* RAS Non-maskable interrupts defined by RNMI spec which is in final stages.
* RAS exception codes will be standardized as part of Priv v1.13 spec by end-of this year

The APEI/CPER as being implemented for RISC-V RAS software stack are independent of RERI and the above mentioned dependencies will be resolved soon.

@hschauhan Please add if I missed anything.

hschauhan, 2023-09-14:
The APEI only captures the information about how events are propagated. APEI doesn't depend on the hardware because I doesn't really deliver the event. The propagation is dependent on hardware but information about it is not. These two things are not same.

dhaval-rivos, 2023-09-14:
Let us say we are proposing to have 2 different types of CPU error records
(we briefly touched on it in our last sync).

   1. Generic Processor Error Section. This one can be independent of RERI
   as information is fairly generic in nature.
   2. Risc-V Processor Error record. This one is RV specific and it will
   take information from RERI and convey it to the OS. So as per my knowledge
   this will have RERI dependency.

There is a follow up to our RAS POC discussion (and Spec changes) will
schedule soon and we can arrive at a conclusion?
=D

avpatel, 2023-09-14:
@dhaval-rivos We can always do APEI related ECRs incrementally. The first set of APEI related ECRs can be very basic having no dependency on HW specs.

dhaval-rivos, 2023-09-14:
Hi Anup,
Yes that is okay too. Another option was to add APEI related details later
in BRS spec once we have conclusions on POC/RERI. I am not sure if it is
critical to have these details in BRS right now (and who would benefit)
when a lot of work is still happening. I think that was the original
question.
=D

andreiw, 2023-10-16:
So it sounds like there may be an AIA dependency (perhaps AIA thus implies compliance to the RAS), but yeah I see that the RAS specs are still in flight so we can hold our horses for sure...

andreiw, 2024-04-11:
Deferring on this.

andreiw, 2026-10-29:
So far the RAS aspects, as they impact UEFI implementations, are being defined in ACPI and UEFI specs. For RISC-V these are still in flight (see https://github.com/riscv-non-isa/riscv-server-platform/issues/111 for details).

I haven't seen any indication that BRS needs to have any RAS-specific language. BRS is also not a server booting spec, and BRS-I and BRS-B approaches (from an ecosystem POV) are diametrical opposites.

I think we can retire this issue.

closed 2026-10-29 by andreiw