riscv-non-isa/riscv-brs #82: Security Requirements - Reference Security model spec

adurbin-rivos, 2023-10-10:
@rsahita had some thoughts he'd help with identifying things to reference in security model spec from BRS spec. While the security model spec isn't ratified, it'd be good to provide pointers in the spec for guidance purposes.

andreiw, 2023-10-18:
We also ought to go over the BBSR (https://developer.arm.com/documentation/den0107/latest/) and see which of these industry-accepted norms and practices are applicable. E.g., requirements on size of variable flash, variable size, access to NV vars, etc.

Some of these follow the BBR trend of repeating upstream spec requirements, of course.

andreiw, 2023-11-16:
https://lists.riscv.org/g/tech-security-model
https://github.com/riscv-non-isa/riscv-security-model

andreiw, 2024-04-11:
As a baseline, I think we can claim success, but I'm going to file a few more issues.

closed 2024-04-11 by andreiw