riscv-non-isa/riscv-brs #89: List the required UEFI boot and runtime services as well as protocols

xiaobo55x, 2023-11-08:
Hi @andreiw, @adurbin-rivos,

Shall we add a section to clearly list all the required and recommended UEFI boot and runtime services as well as related protocols?
With that, users can easily get a checklist when implementing their own FW.

Regards,
Haibo

adurbin-rivos, 2023-11-15:
Section 2.6 in UEFI spec covers the requirements from a UEFI perspective. There is a carveout for EFI_CONFORMANCE_PROFILES_TABLE. We've discussed this in the past, and no one suggested we should deviate from what is in section 2.6.

We didn't want to repeat information in BRS spec that is already covered in existing docs. The things that we call out explicitly are currently captured.

As for a checklist, I think using the test suite would be helpful because those sorts of things would be needed there. We could try to create such a checklist and put it in a non-normative section, but it might be prone to errors.

andreiw, 2023-11-16:
Yeah, 2.6 in the UEFI spec is already exhaustive. If that has errors, they need to be fixed (there were a few broken/misplaced links... they are getting fixed). That's for BRS-I. For BRS-B the EBBR spec is your guide.

closed 2023-11-16 by andreiw