riscv-non-isa/riscv-brs #100: Requirement of UEFI_020

WangYin3, 2024-02-06:
For UEFI_020, the requirement is to "Meet the 3rd Party UEFI Certificate Authority (CA) requirements on UEFI memory mitigations". Does this mean that the firmware needs to meet ALL requirements in “UEFI memory mitigations”? Such as:

CA signing requirements
No Execute (NX) memory protection
Page guards
Pool guards
CPU stack guard
Null pointer protection
Some of these are handled by the application/OS/compiler/CPU. If not, could we list all the necessary requirements in “UEFI memory mitigations”?

andreiw, 2024-02-06:
If you read the https://learn.microsoft.com/en-us/windows-hardware/drivers/bringup/uefi-ca-memory-mitigation-requirements page, you'll note some portions of the text say "must" yet others say "should". The document is sloppy in that it doesn't define must vs should, but it's fair to assume the intention follows something like https://www.ietf.org/rfc/rfc2119.txt. 

I think my interpretation is reasonable, because if you were to treat this list as absolute requirements, then I believe you would be hard pressed to find compliant implementations. Certainly, the reference Tiano code doesn't do page/pool/stack guards (I'm a little skeptical that commercial UEFI impls have all of these, too, just from personal experience...). I believe the "Application of memory mitigation requirements" mostly explains what features the FW could enable if it assumed the alignments the document talks about.

WangYin3, 2024-02-06:
Thanks for quick clarification, so the must requirements are:
1.Section Alignment of the submitted PE file must be aligned with page size. This must be 4 KB, or a larger power of 2 (for example, 64 KB).
2. Section Flags must not combine IMAGE_SCN_MEM_WRITE and IMAGE_SCN_MEM_EXECUTE for any given section. 
I will check if these already covered in SCT test case.

closed 2024-02-29 by WangYin3