riscv-non-isa/riscv-brs #101: Do we need _CCA (AML_010) for all devices?

vlsunil, 2024-02-07:
I think RISC-V is similar to Intel in this regard than ARM. If agreed, it is better to create an ECR to update as  "On Intel **and RISC-V** platforms, if the _CCA object is not supplied, the OSPM will assume the devices are hardware cache coherent. ". With that, the requirement can be removed from BRS?

andreiw, 2024-02-10:
ASWG Mantis 2407 does this already (https://mantis.uefi.org/mantis/view.php?id=2407):

"On Intel and RISC-V platforms, if the \_CCA object is not supplied, the OSPM will assume the devices are hardware cache coherent."

Once the updated ACPI spec is published, the _CCA requirement can be dropped from BRS.

vlsunil, 2024-02-13:
Great!. Thanks Andrei.

andreiw, 2025-01-23:
@vlsunil do we know when the 6.6 spec is going public?

vlsunil, 2025-01-24:
First draft is out for UEFI members review. Unfortunately, there is no published timeline for getting final public version. My guess is it would take at least 2-3 months more.