riscv-non-isa/riscv-brs #111: 8.2.1. Privilege Levels - contradictory information

xypron, 2024-02-13:
The text only describes S, HS, VS as privilege levels under which UEFI binaries must work.

The sentence "In contrast, UEFI drivers, OS loaders and applications must not contain any assumptions of the privilege level at boot time. " would equire EFI binaries to be usable in U-mode and M-mode. This seems unreasonable.

andreiw, 2024-03-02:
The intention was to avoid external UEFI drivers being aware of architectural state, which seems reasonable given these are meant to be portable across multiple UEFI impls, SoC designs etc. I agree the text is sloppy and should be reworded.

X547, 2024-03-03:
> would equire EFI binaries to be usable in U-mode and M-mode.

Assume that I am writing OS UEFI boot loader. How it will be usable in U-mode if operating system kernel to be booted require S-mode?

closed 2024-04-11 by andreiw