riscv-non-isa/riscv-brs #118: Out of scope implementation mandate

darius-bluespec, 2024-02-22:
URT_030 states:

> The UEFI ResetSystem() runtime service MUST be implemented using the
> SBI SRST extension.

However, mandating the implementation is outside the scope of the
specification.

andreiw, 2024-04-11:
To the effect where we are talking about observable behavior - it _is_ in scope. The observable effect of calling UEFI ResetSystem() should be as if the SBI SRST call was done at the end.

andreiw, 2024-04-11:
We can remove the SRST requirement. It doesn't matter how SystemReset is implemented, we're only concerned with it being the only mechanism used by a BRS-I OS.

What about BRS-B? Well it can choose to do whatever, which (most likely) will mean using SRST. A particular BRS-B implementation may impose any number of additional requirements not actually present in the BRS spec.

If we can do this for SRST, why can't we do this for HSM? Because the SBI requirements are effectively the common requirements for BRS-B and BRS-I, and we don't /have/ (or want) another mechanism for hart manipulation.

closed 2024-04-15 by andreiw