riscv-non-isa/riscv-brs #120: Inaccurate description of the SBI

darius-bluespec, 2024-02-22:
Chapter 4 states:

> The supervisor binary interface (SBI) [7] specification defines the
> interface between the supervisor mode and the next higher privilege
> mode.

No it doesn't.  If the SBI were the entirety of the interface between
those modes, the BRS would be unnecessary.

andreiw, 2024-04-11:
I think we're splitting hairs here. It _is_ an interface between S mode and something more privileged, while the BRS is clearly not that in scope. It's unhelpful that the SBI spec itself doesn't have a crisp description of what the spec is (https://github.com/riscv-non-isa/riscv-sbi-doc/blob/master/src/intro.adoc) - it talks about what the SBI spec allows, not what it is.

It sounds like you have a good mental picture of what Chapter 4 should say about SBI, and I can't read your mind. I'm assigning this to you and you can follow up with a PR.

darius-bluespec, 2024-04-18:
It is *an* interface (as you said), not *the* interface (as the
document says).  The former correctly implies that there may be other
interfaces, the latter incorrectly implies it is the sole interface.

Also, the SBI specification says this about itself, which seems fairly
descriptive:

> The SBI allows supervisor-mode (S-mode or VS-mode) software to be
portable across all RISC-V implementations by defining an abstraction
for platform (or hypervisor) specific functionality.

closed 2024-04-18 by andreiw