riscv-non-isa/riscv-brs #125: Inconsistent usage of the word "system"

darius-bluespec, 2024-02-22:
The glossary gives a specific and comprehensive definition for the
word "system", but usage in the specification often differs from this
definition.  The following are some examples.

System used to refer to that which implements the BRS, exclusive of
the operating system, hypervisor, or user application running upon it:

Chapter 2:  "... systems will deliver features beyond those covered by BRS-I"

Section 8.1.1: "Systems compliant to BRS-I ..."

Section 8.3.7: "RISC-V systems compliant to the BRS"

System used to refer to some non-specific portion of the traditionally
hardware portion of the implementation:

Section 7.1: "the surrounding system" as distictly different from "the RISC-V
processor"

andreiw, 2024-04-11:
I hear you on the  7.1 bit. That was borrowed text from the RV spec and I'll clean it up as part of https://github.com/riscv-non-isa/riscv-brs/issues/96 work.

For the other parts, I don't see a problem with the other examples  you've cited. It seems like you have a pretty good idea of how system should and shouldn't be used, so I suggest you draft a PR.

andreiw, 2024-04-11:
I've assigned this item for you. Please draft a PR.

closed 2024-04-18 by andreiw