riscv-non-isa/riscv-brs #127: ACPI Namespace devices for APLIC and PLIC

vlsunil, 2024-03-05:
NOTE: This is still undecided and discussion in progress, just raising the issue here to keep track of it. We can close this if not required.

As part of upstreaming AIA drivers and ACPI support for AIA and PLIC, linux driver model has moved to probe all drivers as platform drivers with FW defining the probe order. Especially APLIC can not be probed early since it needs whole platform device infrastructure to setup MSI device domains. Maintainers are not willing to support core_initcall() for these drivers since they are fragile. So,  creating the dependency between APLIC/PLIC interrupt controllers  and the actual device is the only way to ensure probing order. This would mean something like adding _DEP for devices on APLIC/PLIC. 

So, to add the dependency proposal is to mandate below things in BRS.

1) FW should add namespace devices for APLIC(RSCV0001) and PLIC (RSCV0002) along with MADT. These devices should support _GSB to map correct MADT structure.
2) Regular platform devices should add Interrupt ResourceSource to corresponding APLIC/PLIC for interrupts.
3) PNP devices like 16550A (PNP0501) should add _DEP on corresponding APLIC/PLIC.

vlsunil, 2024-03-05:
Correct. I still think MADT is alone sufficient. Created this issue as per Andrei's suggestion to keep track.

> I guess this is a follow up to your message on this thread. https://lore.kernel.org/lkml/20231226235602.GA1483795@bhelgaas/T/#mcc7c15fcc38042cfbb81f38f7b3220312360acbc
> 
> As per my understanding from the thread, you still prefer the current approach on v3. But if we have to adopt Rafael's suggestion, we have to do as suggested above. Please let me know if I misunderstood something

Correct. I still think MADT is alone sufficient. Created this issue as per Andrei's suggestion to keep track.

andreiw, 2024-04-11:
So for the moment being, no changes are needed to the BRS spec. Let's keep this issue open for now.

andreiw, 2024-04-12:
This is a current work item for the spec - not future.

andreiw, 2024-04-15:
And we're merged.

closed 2024-04-15 by andreiw