riscv-non-isa/riscv-brs #162: Comments on version 0.0.2

ved-rivos, 2024-05-14:
Chapter 5 Some of the requirements are missing the RFC2119 keywords to indicate whether they are mandatory or optional.

ACPI_080 - implies that interrupt controllers may be either PLIC or APLIC. This statement may want to be expanded to include IMSIC and a corresponding RVI ACPI ID provided in section 6.2. In similar vein, section 8.3.6 "the MADT will also contain either PLIC or IMSIC/APLIC structures" would want to be written as "the MADT will also contain either PLIC, APLIC, or IMSIC structures"

andreiw, 2024-05-15:
@vlsunil do we need an IMSIC-specific CID for 6.2?

andreiw, 2024-05-15:
https://github.com/riscv-non-isa/riscv-brs/pull/163 for Chapter 5 improvements

Will send email to the mailing list to discuss the other points

andreiw, 2024-05-16:
#163 is merged.

andreiw, 2024-05-20:
Based onhttps://github.com/riscv-non-isa/riscv-brs/pull/166 and https://github.com/riscv-non-isa/riscv-brs/pull/164, we're good right? @ved-rivos