riscv-non-isa/riscv-brs #168: Many JEP106 manufacturer codes do not fit in 4 bytes

jhauser-us, 2024-05-24:
In Section 7.1, "Type 04 Processor Information", we have:
> The first `DWORD` (offsets 08h-0Bh) is the JEP-106 code for the vendor.

JEP106 defines a manufacturer’s identification code as a variable number of bytes, consisting of zero or more 0x7F bytes followed by a final byte that is not 0x7F.  For all recently registered manufacturers, this sequence is longer than four bytes.  For example, I believe the code for SiFive is a sequence of 10 bytes:  (in hexadecimal) 7F 7F 7F 7F 7F 7F 7F 7F 7F 89.  The document does not currently explain how this is going to be made to fit in the first four bytes of the Processor ID field of a Processor Information structure.

Please note that there is no one standard encoding for compressing the string of 0x7F prefixes, so if such an encoding is intended, it should not be assumed to be obvious common knowledge.

ved-rivos, 2024-05-29:
The Priv. Spec `mvendorid` register implements a compression scheme where bits 6:0 is the ID without the parity and the bits 31:7 represent the number of continuation codes (e.g., 0x509). This specification could just state that `DWORD` holds the value in `mvendorid`.

andreiw, 2024-05-31:
I'll write it out, because the value may be a soc ID - not a vendor id (the soc vendor may differ from hart vendor).

closed 2024-05-31 by andreiw