riscv-non-isa/riscv-brs #177: Question about device properties

vlsunil, 2024-07-18:
It is not clear to me from the description whether _DSD properties should add "riscv-uart-" prefix for each property defined in this section. (REF: https://github.com/riscv-non-isa/riscv-brs/blob/main/acpi-prop.adoc). For ex: for a standard NS16550A UART (RSCV0003), if the FW adds _DSD property with name "rscv-uart-reg-shift", then the standard 16550 driver in UART will not work because it expects the property name to be just "reg-shift". I am sure adding new driver just for this will be NAKd. Do we really need this section?  IMO, _DSD properties should be aligned with DT bindings defined so that same driver works for DT as well.

adurbin-rivos, 2024-07-18:
I think it ca be dropped. Sunil, could you please send a PR for that?

@andreiw - do you know why it would need to be prefixed at all?

andreiw, 2024-07-28:
See 3.1. under https://uefi.org/sites/default/files/resources/DSD%20IMPLEMENTATION%20GUIDE%20VERSION%202.0%20JAN%202021_EDITED_WEBVERSION.html#_device_properties_uuid

Quoting...

While the Key for a Property may be any String, there is a very high probability of name collision. For example, two vendors could use the String "interrupts" as a Key. The problem is that each vendor may have very different use cases for the Value. One use case could be a Boolean value ("yes" means interrupts are supported, "no" means the device must be polled). A second use case could be a list of valid IRQs for the device (Package() {1, 2, 3}).

It is highly recommended that each vendor prefix their Key name with their registered PNP or ACPI vendor ID in order to avoid name collisions. For example, use abcd-interrupts instead of interrupts, especially if the word is known to be in common use in the kernel.

Due to historical usage, Appendix A contains some Keys that do not have any prefix at all. These should not be used in the future; they should be considered deprecated. In Appendix B, prefixed names for these same Keys are defined and should be used henceforth.

The reason these older Key definitions have no prefix is that there was at one time the concept of a global namespace for these Keys, and a process for registering them and defining them. As a practical matter, hardware and software vendors have ignored the whole thing and used the Keys in Appendix A as de facto standards. While we still have the idea of a global namespace for Keys, from now on these must be prefixed with acpi- in the interest of avoiding name collisions. In order to create a Key in the acpi namespace, it must be requested as a merge request to this document via (see https://github.com/ahs3/dsd-guide). These will be reviewed by the UEFI Forum for acceptance.

In Appendix C is a list of the currently known Key prefixes. Any vendor wishing to claim a prefix may do so by requesting a merge request to this document via github (https://github.com/ahs3/dsd-guide, as above). How the vendor chooses to define anything after their prefix is entirely up to them. For example, always assume that abcd-irq and lmno-irq are very different Keys, even though both have irq in the name; vendor abcd and vendor lmno could have radically different semantics for the term irq.

Property names that are not one of those grandfathered in through Appendix A, or defined in Appendix B, or use a prefix not listed in Appendix C must not be used. The use of _DSD Device Properties under those circumstance may have unpredictable outcomes.

andreiw, 2024-07-28:
I think we would need a really good reason to go against the _DSD documentation. I can imagine that we can treat "existing" property names as "grandfathered in", and simply use the prefix for any new properties defined that do not currently exist in Linux/BSD/whatever kernel DT support.

vlsunil, 2024-07-29:
Thanks! Adnrei.  Shouldn't these prefixes be approved by UEFI forum as a merge request to  DSD guide instead of maintaining in BRS spec? Do I miss some thing here?

Also, IIUC the prefixes seem to be for making it unique for a vendor. How does making a standard riscv-uart prefix for all vendors would help?

Wouldn't it be better to simply document standard properties like clock-frequency, reg-shift, fifo-size in this section (specifically for standard NS16550)  and mandate any vendor to use the DSD guide process of reserving the name space to define any new property?

adurbin-rivos, 2024-07-29:
> Wouldn't it be better to simply document standard properties like clock-frequency, reg-shift, fifo-size in this section (specifically for standard NS16550) and mandate any vendor to use the DSD guide process of reserving the name space to define any new property?

That makes sense to me for things that already exist in the driver code. For any properties, it makes sense to follow the guidelines, but existing ones should just work on their own.

vlsunil, 2024-07-30:
> Thanks! Andrei. Shouldn't these prefixes be approved by UEFI forum as a merge request to DSD guide instead of maintaining in BRS spec? Do I miss some thing here?
> 
Okay. Latest DSD guide says,

>  _"In addition to the list below, de facto prefixes exist as a result of vendor ID assignments in the PNP [[PNP_ID]](https://github.com/UEFI/DSD-Guide/blob/main/src/dsd-guide.adoc#PNP_ID) and ACPI [[ACPI_ID]](https://github.com/UEFI/DSD-Guide/blob/main/src/dsd-guide.adoc#ACPI_ID) Registries. I.e. the intc- prefix is owned and managed by Intel Corporation, and mipi- is owned and managed by the MIPI Alliance."_

So, "rscv-" prefix is also granted by default.

Also, there is a note added.

> _There is no need to change existing implementations.
> As a pragmatic matter, the use of prefixes and control of the name space as described here is optional so that we do not disturb existing OS implementations and usage. Existing Device Properties can continue to be used as they are today; there is no requirement to change them to use the new prefix scheme._
> 
> _However, we highly recommend that prefixes be used and that they be registered from this point on for future or new Device Properties. It is ultimately in everyone’s best interest to be clear on who owns what Device Properties, and who is responsible for defining them._
> 
> 
It looks like these clarifications are not yet in a released DSD guide but should be fine since they are approved by ASWG.

Thanks for the discussion. Let me raise a PR  to make UART properties to use global namespace.

vlsunil, 2024-07-30:
Raised PR https://github.com/riscv-non-isa/riscv-brs/pull/188

andreiw, 2024-08-08:
merged, closing this issue

closed 2024-08-08 by andreiw