riscv-non-isa/riscv-brs #180: Machine-level environment is out-of-scope for SBI_060

jhauser-us, 2024-07-30:
It's my understanding that the BRS specification is supposed to constrain a supervisor-level RISC-V execution environment, for an OS or hypervisor.  Specifics of how machine-level hardware and software might provide this S-level environment should be out-of-scope.  (Consider that, for an ISA simulator, there might not even be any M-level environment involved; the S-level environment might be simulated directly, without any M-level RISC-V code.)

Accordingly, SBI_060 can refer to S-level extensions Sscsrind and Ssccfg, but should not mention M-level extensions Smcsrind or Smcdeleg.

andreiw, 2024-08-06:
Your understanding is correct, so SBI_060 indeed should not be mentioning any M-level extensions.

andreiw, 2024-08-08:
https://github.com/riscv-non-isa/riscv-brs/pull/192

closed 2024-08-08 by andreiw