riscv-non-isa/riscv-brs #181: BRS-I acting outside its own scope is a contradiction

jhauser-us, 2024-07-30:
In Section 2.1, "BRS-I Recipe", we have this statement:  "A key tenet of BRS-I is constraining behavior outside the scope of BRS-I."  To me, that reads as an impossible contradiction, saying that recipe BRS-I acts outside the realm in which it may act.

avpatel, 2024-10-24:
I believe this is already taken care hence closing.

jhauser-us, 2024-12-03:
The consensus of the Architecture Review Committee is that this sentence reads as self-contradictory:  "A key tenet of BRS-I is constraining behavior outside the scope of BRS-I."  Please find different words to convey what you really mean.

vlsunil, 2024-12-03:
@jhauser-us : 

"A key tenet of BRS-I is to ensure there is no unexpected behavior in a generic OS even when there are  features implemented in the system beyond the scope of BRS-I."

Is it acceptable?

kasanovic, 2024-12-04:
This is still contradictory.  I believe the previous paragraph provides most of the intended mandate.  I'd propose dropping the problematic paragraph, and instead adding a new final sentence on the previous paragraph that states: "Any additional firmware features that cause anomalous and unexpected behavior must be disabled by default, and only enabled by opt-in."

vlsunil, 2024-12-04:
Sure. Thanks!

closed 2024-12-06 by avpatel