riscv-non-isa/riscv-brs #183: Secondary harts should be usable after boot

jhauser-us, 2024-07-30:
UEFI_050 says that, after the `EFI_EVENT_GROUP_READY_TO_BOOT` event, all secondary harts should remain "offline from that point on".  It's a little hard to believe that an OS/hypervisor is advised not to make use of any secondary harts after the system boots, so perhaps this requirement needs re-wording.

andreiw, 2024-08-06:
This is in regards to -firmware manipulation-. Once EFI_EVENT_GROUP_READY_TO_BOOT  the firmware should not be, on its own accord, making any use of the secondaries. A boot loader or application program can of course do whatever it needs to do, and the OS of course can fire these up. But the state of the secondaries when the OS loader runs is quiesced, which in the context of more advanced chips may well mean "powered off and not running any code".

andreiw, 2024-08-13:
Please close if no further action is desired.

jhauser-us, 2024-09-27:
@andreiw:  Firmware authors are not the only audience for the BRS spec.  Another audience is OS authors whose software is a client of the SMBIOS, ACPI, SBI, and UEFI.  They need to know what the BRS spec does and doesn't promise for them.

If we take the second part of UEFI_050 literally, then it does say that an OS can't touch "all secondary harts ... from that point on".  Because if an OS were to use them, they wouldn't be "offline" any more, violating what the rule plainly says.  All I'm asking is that the wording be adjusted to convey better what you wrote above.

ved-rivos, 2024-09-28:
It may want be reworded as follows:
All hart manipulation internal to a firmware implementation SHOULD be done before completion of the EFI_EVENT_GROUP_READY_TO_BOOT event. Firmware MUST place all secondary harts in an offline state before completion of the EFI_EVENT_GROUP_READY_TO_BOOT event.

The note under the requirement could state something to the effect that "The OS loader and/or the OS may resume the secondary harts, if required, as part of their boot and join sequence."

This avoids the connotation that they remain offline forever from that point onward.