riscv-non-isa/riscv-brs #184: AML_070 doesn't require implementation of Get/Set Real Time?

jhauser-us, 2024-07-30:
For AML_070, when a system must implement the Time and Alarm Device, is it acceptable not to implement the Get/Set Real Time methods (`_GRT`, `_SRT`), and have `_GCP` return a value with bit 2 = 0?

andreiw, 2024-08-06:
A TAD device is an appropriate choice for RTC when the RTC access needs to interlock with some other OS code, because the RTC shares a bus with some other device of interest that is being managed by an OS.

If the RTC is implemented via TAD, then it *cannot* be implemented via UEFI RT (the URT_020 rule).

The AML_070 rule was specifically around implementing time access, so this rule implies _GRT and _SRT are implemented and _GCP returns bit 2 = 1.

jhauser-us, 2024-08-06:
> this rule implies _GRT and _SRT are implemented and _GCP returns bit 2 = 1.

That's not good enough.

The point of this issue is that AML_070 does not *say* that `_GRT` and `_SRT` must be implemented.  It's unwise to assume readers will always understand your motivations and therefore always deduce the correct implications.  Worse, you're making it too easy for implementors to purposefully skirt the rule with the excuse that "it doesn't actually say that".

Is there some objection to making AML_070 clearer about the intention?

andreiw, 2024-08-08:
No objections. Will rework the text for clarity.

andreiw, 2024-08-08:
https://github.com/riscv-non-isa/riscv-brs/pull/193

closed 2024-08-13 by andreiw