riscv-non-isa/riscv-brs #185: Clarify when multiple SMBIOS structures of same type are required

jhauser-us, 2024-07-30:
SMBIOS_040 says, "A Port Connector Information (Type 08) SHOULD be implemented."  Just one?  Or is the intention that this structure be implemented for each relevant port?

SMBIOS_110 says, "An Onboard Devices Extended Information (Type 41) structure SHOULD be implemented."  Again, just one?

SMBIOS_140 says, "A Processor Additional Information (Type 44) structure MUST be implemented."  Just one?  Or is the intention to require one for each RISC-V hart?

SMBIOS_150 says, "A Firmware Inventory Information (Type 45) structure SHOULD be implemented."  Just one?  The SMBIOS spec says, "One Type 45 structure is provided for each firmware component."  So maybe SMBIOS_150 would be better to say, "... SHOULD be implemented for each firmware component".

Similarly, for SMBIOS_050, instead of "A System Slots (Type 09) structure MUST be implemented, ...", perhaps it would be better to say "System Slot (Type 09) structures MUST be implemented, ...."  The SMBIOS spec labels this structure as both singular, "System Slot", and plural, "System Slots", but it also clearly says, "One structure is provided for each slot in the system."

andreiw, 2024-08-06:
With any of these, the SMBIOS spec applies. The purpose of the /SHOULD/ text is really to call attention that unless there's a good reason not to, these structures should be implemented as per SMBIOS spec.

Type 44 guidance is provided by the BRS. A Type 44 structure needs to be provided for every hart meeting [[hart]](https://github.com/riscv-non-isa/riscv-brs/blob/main/smbios.adoc#hart) requirements.

andreiw, 2024-08-08:
Systems slots was removed as part of cleanup some time after the document was submitted to ARC.

Working on removing singular for the other ones.

closed 2024-08-08 by andreiw