riscv-non-isa/riscv-brs #198: Despite claim, no properties are defined in the rscv- namespace

jhauser-us, 2024-09-28:
Section 6.3, "RVI-specific ACPI Device Properties", says, "This section defines the `_DSD` device properties in the `rscv-` namespace."  However, the only properties actually mentioned in this section are for UART devices, and if I'm not mistaken, these are *not* defined in the `rscv-` namespace.  (Right?)

In short, I believe this section says it defines properties in the `rscv-` namespace but then doesn't do so.

vlsunil, 2024-10-11:
You are right. I think this is a place holder for any property to be defined under rscv- namespace. The DSD guide (https://github.com/UEFI/DSD-Guide/blob/main/src/dsd-guide.adoc#appendix-c-known-device-property-prefixes) allows to use vendor ACPI ID as the prefix for the property but vendor needs to manage those properties. For RISC-V, ACPI ID registered is RSCV. Hence, any one in future wants to define and use a new property under rscv- namespace, should register here first. I think that is the idea to have this section.

andreiw, 2024-11-06:
We had some properties here initially, then - we didn't. Nevertheless, it makes sense to stake a claim to the rscv- namespace now.

andreiw, 2024-11-06:
Marking this as won't fix given last two comments. If you're okay with that, please close the ticket @jhauser-us.

jhauser-us, 2024-11-07:
The right response is not to leave the document making an obviously false statement.  If you want to stake the claim to the `rscv-` namespace, then change the text to say that.

jhauser-us, 2024-12-03:
It is the consensus of the Architecture Review Committee that Section 6.3, "RVI-specific ACPI Device Properties", needs a non-normative note clarifying that the table of `rscv-` names is currently empty.  And this paragraph:
> Request for new property names in the `rscv-` namespace should be made as a git pull request to below table in this document.

should be made non-normative, preferably moved into the same non-normative note.

In my opinion, it makes more sense for this non-normative note to *follow* the table, which would require "below table" be changed appropriately.  (Note, by the way, that the words *above* and *below* are not adjectives, so it's not gramatically correct to say "above table" or "below table".)

vlsunil, 2024-12-03:
Thanks @jhauser-us . Let me update as per your suggestion.

closed 2024-12-06 by avpatel