riscv-non-isa/riscv-brs #200: Give a reason for UIO_010

jhauser-us, 2024-10-07:
The Architecture Review Committee would like the BRS document to include some non-normative justification for UIO_010, which obligates the UEFI firmware to initialize all PCIe root complexes.

ved-rivos, 2024-10-31:
The non-normative justification slightly hints at the fact that this is a stronger requirement than PCI BIOS envisioned. It will be useful to augment that justification to state why PCIe resource allocation and host bridge configurations are not really standardized across platforms and without the firmware initializing the root complexes it may lead to unexpected behaviors.

avpatel, 2024-11-19:
The commit https://github.com/riscv-non-isa/riscv-brs/commit/f967ab2d46bf07865d235f058353e384ba28d9ec addresses this issue hence closing it.

closed 2024-11-19 by avpatel