riscv-non-isa/riscv-brs #219: Split up UEFI_050

andreiw, 2025-01-23:
Today this says "All hart manipulation internal to a firmware implementation SHOULD be done before completion of the EFI_EVENT_GROUP_READY_TO_BOOT event. Firmware MUST place all secondary harts in an offline state before completion of the EFI_EVENT_GROUP_READY_TO_BOOT event."

...the problem is that it's mixing together both a MUST and a SHOULD all in one rule. So, if someone was satisfying the MUST, but for whatever reason not the SHOULD, it could become confusing to reason about the rule... 

I suggest splitting these out.

UEFI_050 - Firmware MUST place all secondary harts in an offline state before completion of the EFI_EVENT_GROUP_READY_TO_BOOT event.
UEFI_050 - All hart manipulation internal to a firmware implementation SHOULD be done before completion of the EFI_EVENT_GROUP_READY_TO_BOOT event.

andreiw, 2026-10-12:
@vlsunil Is this still interesting?

vlsunil, 2026-10-15:
Makes sense. Could you raise a PR?

vlsunil, 2026-11-02:
PR merged

closed 2026-11-02 by vlsunil