riscv-non-isa/riscv-brs #220: Flesh out UEFI_020

andreiw, 2025-01-23:
Today this just points to the 3rd Party UEFI Certificate Authority (CA) requirements page on Microsoft's domain, which is a bit annoying, since:
- The document (https://learn.microsoft.com/en-us/windows-hardware/drivers/bringup/uefi-ca-memory-mitigation-requirements) is not a spec. It's an article. There's no version. It can be arbitrarily pulled, reworked, or change location.
- The article is really talking about signing requirements (for OS loaders). Sure, it also talks about some possible /applications/ of those requirements, that is what firmware could do if all binaries followed these rules, but most of the content on the page is irrelevant.

Suggestion is to drop the link entirely, and instead flesh out the actual requirements, which appear to be:
- Section Alignment of DXE-mode modules (including runtime drivers) must be aligned with page size. This must be 4KB. It can be larger.
- Section Flags must not combine IMAGE_SCN_MEM_WRITE and IMAGE_SCN_MEM_EXECUTE for any given section.
- EFI_MEMORY_ATTRIBUTE_PROTOCOL (we already have this covered under a different rule)

As description we SHOULD say something along the lines of not breaking OS DEP/NX when mapping UEFI RT ranges. It should be clear that this is not a "nice to have" for testing/security reasons, but instead can have direct impact on the OS after boot.

Btw here's the Arm BBR wording. Btw I wasn't able to find anything about 64k alignment for DXE_RUNTIME_DRIVER. Maybe I didn't look very well.

6.3.3.3. PE/COFF image
The SectionAlignment and FileAlignment fields in Microsoft PE Format must contain the value of
at least 4KiB. Higher values are possible. For example, UEFI specification requires
DXE_RUNTIME_DRIVER modules to have the 64KiB granular memory type.
Modules that execute in place, for example SEC, PEI_CORE or PEIM type UEFI/PI modules, are
exempt from this requirement. For these modules, any power-of-2 value of 32 bytes or higher is
possible if the section alignment and file alignment are equal.
PE/COFF images whose section alignment is at least 4KiB must not contain any sections that have
both the IMAGE_SCN_MEM_WRITE and IMAGE_SCN_MEM_EXECUTE attributes set.

andreiw, 2026-10-12:
@vlsunil is this still interesting?

vlsunil, 2026-10-15:
Yes, not a good idea to refer to pages which are not versioned. I don't understand how UEFI spec specifies 64kib in general for ARM for runtime services. I can see it is applicable only if OS uses 64k page size. Anyway, we don't need that for RISC-V. Will you be able to raise PR?

andreiw, 2026-11-04:
https://github.com/riscv-non-isa/riscv-brs/pull/310

closed 2026-11-05 by vlsunil