riscv-non-isa/riscv-brs #221: UEFI_030 non-normative text unnecessarily constraining?

andreiw, 2025-01-23:
The cursive non-normative text under UEFI_030 talks about satp configuration mode but the spec talks about observed behavior, not about implementation details.

Satp is a means to an end, but it relies on DRAM being actually mappable 1:1.  Unfortunate system design choices, like picking bit 47 on an Sv48 system, would mean 1:1 is not possible with satp, but an implementation could use hgatp just fine (running most of UEFI in VS mode). All it needs to do is drop out of VS mode during ExitBootServices to not freak out the OS and it could still meet this spec.

So I'd either drop the "satp.MODE must not be set to Bare" text or reword it to make it clear this is an example of "enabling paged virtual memory", not an rule to follow.

vlsunil, 2025-02-06:
Bare mode is explicitly avoided for enabling protection. This was added after extensive discussion with ARC as well. REF: https://github.com/riscv-non-isa/riscv-brs/issues/182#issuecomment-2466671916

andreiw, 2025-02-13:
I know why we avoid bare mode - after all, I originated the requirement. The problem I see is forcing a specific implementation. The goal is MMU enforcement of access rights, and I just don't see clear value of defining the way to accomplish that. It could be second stage, or even something crazy like PMP. The mechanism shouldn't matter. We should be talking about observable behavior.

vlsunil, 2025-03-19:
Okay.. It is anyway non-normative text. Let me drop this before going for public review. Thanks!