riscv-non-isa/riscv-brs #224: Move/rework URT_060

andreiw, 2025-01-26:
URT_060 should move into the BRS-I Security Requirements, which should go after the RT section.

Moreover, URT_060 should be extended to include this rule subrule - "UEFI variables must be saved to dedicated protectws storage which is not directly accessible by the operating system or other devices "

Well, URT_060 could then be split up. There's no value to have a single rule with 10 items to it.

vlsunil, 2025-02-06:
Since ARC review is completed, we will consider adding new subrule/requirement for the next version of the spec. However, I will make cosmetic change like moving and splitting the requirement to multiple in 1.0 itself. Please create a new enhancement issue to add the new subrule for next version.

closed 2025-02-07 by avpatel

andreiw, 2026-10-29:
"UEFI variables must be saved to dedicated protected storage which is not directly accessible by the operating system or other devices" suggestion is already implemented as SEC_070 in server platform 1.0 - " The platform and firmware MUST back the UEFI Authenticated Variables implementation with a mechanism that cannot be accessed or tampered by an unauthorized software or hardware agent."